klavinslab/aquarium-bugs #6: Planner view

dvnstrcklnd, 2017-07-11:
In the planner view, can the samples be listed as "#{sample.id}: #{sample.name) in #{sample.container_type.name}"? This would help clarify how the sample is progressing through the transformations and changes state. Right now, seeing it listed exactly the same way makes it difficult to see what is happening.

klavins, 2017-07-11:
Isn't the fourth column of the "Inputs, Outputs, and Parameters" box the container type? Or do you mean somewhere else?

dvnstrcklnd, 2017-07-11:
I mean in the tree view of the workflow.

klavins, 2017-07-11:
In that view, the container type is shown on the left when the sample input is selected. Since it is an input box, I didn't know where to put the container type while also keeping the tree small (they can get quite large). Ideas?

dvnstrcklnd, 2017-07-12:
[image: Inline image 1]

dvnstrcklnd, 2017-07-12:
[image]

klavins, 2017-07-12:
ok.

klavins, 2017-07-12:
[image]

How does this look?

dvnstrcklnd, 2017-07-12:
Admittedly I was on an iPhone, but it took me a few minutes to notice what
you had done. Instead of right justifying the container, how about

9964: L1-AZI1-fdf <span style=color: blue>(Primer Aliquot)</span>

all left justified.

It's less of an issue in this workflow, but in the workflow Garrett and I
were working on today there were a lot of operations that had the same
input and output. It makes it hard to see what's happening to the sample.

klavins, 2017-07-12:
The problem is that left justifying it gets in the way of the user's input. In your example "9964: L1-AZI1-fdf" is typed in by the user. I don't want them to have to type "9964: L1-AZI1-fdf (Primer Aliquot).

dvnstrcklnd, 2017-07-12:
But that gets populated by autocomplete, right? I think I've done autocompletes that join two or more tables, but maybe too fussy for this. What you have is fine.

closed 2017-07-12 by klavins