klavinslab/aquarium-bugs #8: Extract Gel Slice errors out due issues getting uploads from show block

jvrana, 2017-07-12:
Error in:
`Upload.find(gel_uploads["134742"][:my_gel_pic][:id])`
- This should be resolved before launch

klavins, 2017-07-14:
This one is taking longer than I though. I hope there is a work around until I fix it. For example, I think you mentioned finding the Uploads associated with the job and using that?

jvrana, 2017-07-14:
I fixed the Extract Gel protocol for now with some ugly-ish code but its working!

jvrana, 2017-07-14:
Once we get the uploads returned in the show block, I'll got back and fix it proper. But the protocol should run without a hitch (I hope). Uploads are easily gotten via the job-id

klavins, 2017-07-14:
I've got it fixed now, but it won't be pushed to the server until tonight.

klavins, 2017-07-14:
Example code:

```
class Protocol

  def main

    data = show do
        title "Upload Something"
        upload var: "images"
    end
    
    uploads = data[:images].collect { |u| Upload.find u[:id] } if data[:images]
    uploads = [] if !data[:images]
    
    show do
        title "Results"
        note "Thanks"
        uploads.each do |u|
          note "upload: #{u.id}"
        end
    end
    
    {}
    
  end

end
```

closed 2017-07-14 by klavins