klavinslab/aquarium-bugs #12: Can't select newer collection items in planner

gasnew, 2017-07-13:
For example, Check Divided Yeast Plate, when planned, will always use the oldest available divided yeast plate as an input, regardless of what is selected while planning.

For example, try replanning plan 16096 on the nursery and selecting the newer of the two divided yeast plates. Once the plan is launched, it will always point to the old one (137452).

This seems to only affect collections, and the row and column from the selected plate carry over into the operation , rather than the row and column from the plate that the operation points to.

klavins, 2017-07-14:
I think I fixed this. Please check.

gasnew, 2017-07-14:
Could you please push this to nursery so I can test it there?

closed 2017-07-14 by klavins

gasnew, 2017-07-14:
Ok, looks fixed!