klavinslab/aquarium-bugs #9: Order Primer precondition gives an error because debug is not defined

klavins, 2017-07-13:
I don't think the "debug" method is available in the precondition code.

undefined local variable or method `debug' for #<Operation:0x007fd1fba75a88>Did you mean?  debugger: /Users/ericklavins/.rvm/gems/ruby-2.3.0/gems/activemodel-3.2.19/lib/active_model/attribute_methods.rb:407:in `method_missing'

gasnew, 2017-07-13:
That's odd. I've run that precondition successfully before (and tested it in Planner). Same for other operation types, like Yeast Lysate.

Though, I have had trouble developing preconditions. It's hard to determine the cause of the issue with the only error message I've seen: "One or more preconditions could not be evaluated," but I figured the cause for the error was on my end. Is there any reason "debug" would sometimes be defined and other times not be?

klavins, 2017-07-13:
Note that if the precondition fails, the plan containing its operation will get the error. That's where I got the error above. Try it on the nursery server. I tried it on my laptop after importing the workflow.

gasnew, 2017-07-13:
Ahh, I see what you're saying. I'll close this then. The precondition probably doesn't need to know whether we're debugging because we can opt not to use the precondition while testing.

closed 2017-07-13 by gasnew