klavinslab/aquarium-bugs #25: Nice to add object_types to Data column in job/operation view of the Manager

jvrana, 2017-07-14:
When you click on a job in the Manager page, it displays a nice data table, that is super useful
[image]

I think if we also added the object_type from the field value and from the item (if there's an associated item) it'd be faster to debug plans and plan issues, especially with operations that change object types. Something like:

**Data:**
out: Fragment: PS_yeGFP **as a Stripwell**
in: Forward Primer : PS_F_assembly **as a Primer Aliquot** ( 68318 **Primer Aliquot** )
  Reverse Primer : yeGFP-R  **as a Primer Aliquot**  ( 13703 **Primer Aliquot** )
  Template : mScarlet-C **as a Fragment Stock** ( 110602 **Fragment Stock** )

jvrana, 2017-07-14:
This is also similar to #24

klavins, 2017-07-20:
Ok, I've added this info, but it looks a bit verbose sometimes:
  Primer Aliquot : 20298: optimization-test-fragment-reverse primer as Primer Aliquot ( 99463 )

closed 2017-07-23 by klavins