klavinslab/aquarium-bugs #28: Unused items are being generated

gasnew, 2017-07-14:
I don't know how to reproduce this, but item IDs are increasing faster than expected, and searching for some IDs between existing item IDs yields no result. For example, both items 98313 and 99943 exist, but 
99900 does not.

klavins, 2017-07-14:
I need a minrep.

gasnew, 2017-07-14:
Testing operations in developer causes this bug.

I ran an Order Primer plan, tested Order Primer in developer with 5 test operations, and then ran another Order Primer plan. The Lyophilized primer IDs from the first and the second plans were 137582 and 137588 respectively.

klavins, 2017-07-14:
Ugh, this is bad. Let's not run testing in production mode until I figure this out. #

klavins, 2017-07-14:
Oh wait, developer is not actually making those items, it is just using up those item numbers, right? That's actually the expected behavior. The only issue I see with this is that you'll use up item numbers if you run tests in development. But that isn't that big of a deal, right?

gasnew, 2017-07-14:
That's right; testing doesn't seem to be making items. But I was made aware of this because Michelle noticed it and was bothered by it. I thought she meant she was seeing item IDs jump within a single job, but that seems unlikely if testing is the cause.

Still, I think it would be better for us not to have item numbers that don't point to items.

klavins, 2017-07-14:
That's true. Unfortunately, the way testing works is to make all the MySQL rows the test needs and then rollback the entire transaction.  The way the MySQL rolls back does not reset the auto_increment columns because if it did, you could get race conditions with someone else using the database at the same time. You get new item numbers as a result. To fix this issue I would have to either (a) rewrite the test procedure not to use a transaction/rollback, which seems really difficult or (b) not use Item.id as the item's id, but rather have separate "actual_item_id" field. In fact, the latter is best practice in MySQL, which a practice I didn't know about four years ago when I started this project. Since the item id's are littered throughout the code, it seems like a bad idea to try to refactor this at the present time. Since neither (a) nor (b) is a good option, we are left with: don't run tests in production mode if you can help it.

closed 2017-07-14 by klavins

gasnew, 2017-07-14:
That makes a lot of sense. I'll keep that MySQL pointer in mind for when I make my own applications in the future.

Thanks for the explanation!