klavinslab/aquarium-bugs #29: Nice to Link inventory to existing operations in a plan

jvrana, 2017-07-14:
Maybe a behemoth request, but it would be extremely useful feature if, in addition to the existing inventory that shows up on the left hand side of the Planner page, that operations that would produce that item in the planner would also show up, clicking on it would create a wire from the output of that operation to the input of your current selected operation.

Even more useful would be if you could link (create wires from) operations in other plans that already exist. This would allow you to say, create a Purify Gel Slice in one plan, submit it, then create a new plan for Gibson Assembly that references the output of the Purify Gel Slice in a different plan.

I tried to make a pic of what I mean:

[image]

Basically on the left side, it would list all operations that:

- have an output field value with the allowable field value equivalent to the field value you are selecting in the planner
- and are active or in planning

Clicking on that operation, it will make a wire between the two field values, satisfying the condition that its ready to launch.

jvrana, 2017-07-14:
This is kind of in reference to #27. You'd be able to choose how you consolidate like operations, especially if the "Auto" preferentially chose existing operations in your plan to link.

Currently, its not really possible to link a Fragment that is under construction to the input of a new plan. You have to wait either:
- for it to be done and then create your plan
- cancel your current plan that includes that fragment and create a new plan
- or create a plan that includes a replicate Fragment workflow

gasnew, 2017-07-14:
I like this a lot.

This is very similar to an idea @ayeshasal and I had for handling operations with array outputs. An example where Send to Sequencing has an array output:
Plan goal 1: Plans from Glycerol Stock all the way back to Primer Orders
Plan goal 2: Plans from Glycerol Stock back to Send to Sequencing (the same operation instance being planned in "Plan goal 1")
Plan goal 3: Same idea as "Plan goal 2"

Once the user has planned "Plan goal 1" and picked three sequencing primers, they should just have to click Add Goal and Auto for each subsequent goal to submit the three sequencing reactions for a single stock.

klavins, 2017-07-15:
Yeah, all this seems good, but difficult. Two issues are doing this is mainly limitation of the UI that I've made that does trees. Better would be a Simulink style system where you could drag operations around and wire them however you wanted. The other issue is that the current client side planner is poorly written and way too coupled with the UI, making these kinds of things difficult. Nevertheless, I wholeheartedly agree it needs to be done eventually. 

In any case, related to issue #27 I have added a backend optimization that removes redundant operations from your plan.

ayeshasal, 2017-08-10:
i'm closing this because i think the new planner solves this? re-open though if y'all think it doesn't

closed 2017-08-10 by ayeshasal