klavinslab/aquarium-bugs #35: Make PCR Fragment fails if a primer temperature is zero.

klavins, 2017-07-15:
In the off chance that someone forgets to put in their annealing temperature for their primer.

jvrana, 2017-07-17:
I have some suggested code for this that I wrote for the TC Culture workflow.

The following precondition will hold back PCR that have primers with invalid Annealing Temperatures and display a warning to the operation and plan. Plan warnings are combined from other operations if they already exist, and operation specific warnings are cleared if they pass (i.e. if user updates the T Anneal). Once conditions are met, messages are removed and precondition is true.

```ruby
def validate_property sample, field_name
  validation_block = Proc.new
  property = sample.properties[field_name]
  if property.nil?
    raise "Could not find property #{field_name}"
  end
  validation_block.call(sample, property)
end

# Appends associates msgs to plan and operation
def precondition_warnings op, msgs, key=:precondition_warnings, sep=";"
  plan = op.plan
  if plan
    plan_msg = op.plan.get key
    plan_msgs = plan_msg.split(sep).map { |x| x.strip } if plan_msg
  end
  plan_msgs ||= []
  op_msgs = []

  msgs.each { |valid, m|
    if valid
      plan_msgs.delete(m)
    else
      plan_msgs << m
      op_msgs << m
    end
  }
  op_msgs.uniq!
  plan_msgs.uniq!

  op.associate key, op_msgs.join(sep + " ")
  op.plan.associate key, plan_msgs.join(sep + " ") if op.plan
end

def precondition(op)
  ready = true
  msgs = []

  # Validate sample, valid_block, valid_message
  ["Forward Primer", "Reverse Primer"].each do |n|
    sample = op.input(n).sample

    msgs << validate_property(sample, "T Anneal") { |s, property|
      validator = Proc.new { |p| p.to_f.between?(0.01, 100) }
      msg = "T Anneal #{property} for Primer \"#{s.name}\" is invalid"
      [validator.call(property), msg]
    }
  end
  msgs.select! { |valid, m| !valid }
  msgs.compact!
  precondition_warnings op, msgs
  ready = false if msgs.any?
  ready
end
```

gasnew, 2017-07-21:
That's nifty! I'll give it a shot in PCR.

I think, in the future, this sort of checking (property validation) should be done before the plan is allowed to launch, like a "compile-time" script of some sort?

closed 2017-07-21 by gasnew