klavinslab/aquarium-bugs #37: Nice to have the plan optimization optional

jvrana, 2017-07-16:
I like the new plan optimizer, however I think having the new plan optimization feature optional would be useful (as a "Optimize Plan" toggle switch in the planner?). The optimizer is making it difficult to have Aquarium do exactly what I detail in the plan, which was a powerful feature in itself. Here's a pic of a situation that I think cloning users might similarly run into:

[image]

In the above situation, I intend to create two Minipreps from two different TB Overnights. Then I want to create two different PCR fragments from each miniprep (maybe I want to see if either of the colonies produce a 'good' band).  However, the plan optimization is removing the PCR, meaning in the plan, I created a miniprep, presumably for no reason.

Other use cases happen alot in my mammalian cell workflows, almost in every plan. There's a workaround in which I will be forced to submit single goal plans, which can get cumbersome.

However, I little toggle switch would allow me to do exactly what I detailed in the plan, like making 5 PCR fragments from 5 different Miniprep all of the same sample.

klavins, 2017-07-16:
Ok, I'll add a switch so you can decide if you want to use it or not.

jvrana, 2017-07-16:
Awesome

jvrana, 2017-07-16:
I also imagine that users may not realize what the toggle switch does (bc they are users), so maybe if there are optimizations, also displaying a little note under the Optimizations thing on the left telling them that they can turn off the optimizations using the toggle switch and clicking Launch would be useful as well.

Also, colorizing the Optimization messages so they pop a little more since they look pretty inert and blend in (maybe green because optimizations are good?)

klavins, 2017-07-16:
Switch added. Its off by default. It has a tool tip. And if selected, then messages show up on the left telling you what, if anything, happened. 
[image]

closed 2017-07-16 by klavins