klavinslab/aquarium-bugs #41: Manager sometimes miscounts waiting and pending operations

gasnew, 2017-07-18:
We discovered this with Make Glycerol Stock, where inputs and preconditions were met, but the Manager interface still counted the pending MGSs as waiting. Selecting pending MGSs did, however, reveal the pending MGSs. Here's an example of the bug where some MGS operations are waiting and others are pending.
[image]
and after pending MGSs have been scheduled
[image]

bjkeller, 2017-09-07:
Is this a protocol issue, or is it something in the `Protocol` class?

cpc3, 2017-09-08:
This is becoming a bigger issue.  We have about 4-5 protocols that are doing this today and it's hard to keep track of/to click on every single protocol to see if something is "pending" even though it says there's none.

bjkeller, 2017-09-08:
@klavins There seems to be an update issue for the `numbers` array when the manager view [operations table rows are created](https://github.com/klavinslab/aquarium/blob/master/app/views/operations/index.html.erb#L44).

When @gasnew updated the protocol preconditions to include the wait, the managers noticed that the counts in the table were not updated even though clicking on the table entries showed that the status of the operations had changed.

The current status is that the protocol preconditions have been rolled back.

klavins, 2017-09-08:
A minrep I can run on my laptop would be very useful.

bjkeller, 2017-09-08:
@gasnew can you provide @klavins with the preconditions and a scenario?

gasnew, 2017-09-15:
I've made a fix for this that we should definitely merge before we update everything on Monday.

klavins, 2017-09-15:
See my comment to your pull request. Still some work to do.

klavins, 2017-09-19:
Garrett's fix for this was merged from pull request #264

closed 2017-09-19 by klavins