klavinslab/aquarium-bugs #48: "Cost Undetermined" error in Nursery

ayeshasal, 2017-07-20:
i'm porting the media protocols, but when i try to plan a media protocol, i get this error:  "Cost Undetermined: Make Liquid/Agar Media operation: Child sample not specified for output 'Media'." not sure what this means, exactly. 

if someone could point me to where in the code the problem might be, i could look into it (i saw that _launch_plan.html.erb invokes plan.cost.error to display the error, but when i went to the definition of "cost," i ended up in cost_model.rb--we aren't using that anymore though, right?). thanks!

klavins, 2017-07-23:
Which one doesn't work? I tried Prepare YEB liquid medium and it planned just fine.

Note that the error has something to do with there not being a sample type associated with one of the inputs or outputs.

ayeshasal, 2017-07-25:
it's the 'Make Liquid/Agar Media' operation, and there is a sample type associated with the output (there is no input except a parameter). could the issue be that there are multiple possible output sample types/object types (e.g., 200 mL liquid, 200 mL  agar, etc.)?

closed 2017-08-03 by ayeshasal