klavinslab/aquarium-bugs #53: Transform cells does not allow Plasmid or Golden Gate inputs

wrightrc, 2017-07-21:
Only Gibson Reaction Result and Ligation Product are available in the left side menu of the planner.

gasnew, 2017-07-21:
Try using Transform Cells from Stripwell. That should allow you to wire to Golden Gate.

wrightrc, 2017-07-21:
That works for golden gate but I still have no way to transform item 99195 (a plasmid stock) into E. coli in order to propagate and make a new plasmid stock.

gasnew, 2017-07-21:
That makes sense. @jvrana is on this one!

jvrana, 2017-07-21:
Fixed! Plasmid Stocks and 1 ng/ul Plasmid stocks are now options! Plasmid stocks will be diluted to 1 ng/ul Plasmid Stocks in the protocol and new item created. The operation will display a warning if this happens and will automatically rewire the input to the new diluted stock. 

Which means, if this happens, the diluted stock will now appear as the input in the planner instead of the plasmid stock you used as an input. This can be changed if its confusing for users.

closed 2017-07-21 by jvrana