klavinslab/aquarium-bugs #59: It would be nice to have an interface for making collections

gasnew, 2017-07-25:
This would be helpful for importing collections to the lab and for correcting for protocol errors.

frba, 2019-03-27:
It would be great if we could change/edit the name of the columns and rows, to match with the names in plates. Like in the image below:

[image]

[image]

bjkeller, 2019-03-27:
Is this just for display purposes, or do you want to be able to use the information within a protocol?

frba, 2019-03-27:
At first, I just thought about using it as a display. Because I was adding new samples to the database and I was confused to correlate well A1 with the 0,0 position and so on (A2 -> 0,1).
However, I believe having this information in the protocols would be useful.

bjkeller, 2019-03-27:
OK, sorry I was confused.

Would it be sufficient for you to have these be plate coordinates? e.g., A01, B12...

frba, 2019-03-27:
> Would it be sufficient for you to have these be plate coordinates? e.g., A01, B12...
>  **Yes, that is what I need.**
> In other words, is it correct that you don't need to have arbitrary row/column names? **No**

bjkeller, 2019-03-27:
I'm going to close this issue, because technically it is done. But added corresponding issue on Aquarium repo (https://github.com/klavinslab/aquarium/issues/345) for @frba's request

closed 2019-03-27 by bjkeller