klavinslab/aquarium-bugs #66: Collection rows and columns are sometimes not assigned in field value once plan is launched

gasnew, 2017-07-27:
This occurs when a user doesn't click on the row + column they want to use as an input to an operation. See the following Check Divided Yeast Plate operations:
- [Plan 1058](http://52.27.43.242/launcher?plan_id=1058#): I used Alberto's plate and didn't click on [0, 3]
- [Plan 1060](http://52.27.43.242/launcher?plan_id=1060#): I used Alberto's plate and clicked on [0, 3]

[image]

This bug also applies to collections in which there is only one of the desired sample.

gasnew, 2017-07-28:
This also seems to occur when copying a plan that takes a collection part as an input to an operation.

klavins, 2017-08-04:
Hopefully this will be fixed by the new planner GUI. It chooses a row and column by default. It also has a validation step that doesn't let you launch if there is now row and column.

closed 2017-08-08 by klavins