klavinslab/aquarium-bugs #67: Unable to add images to protocols

dvnstrcklnd, 2017-07-28:
I can't seem to get any images to show up. I've tried .pdf, .png, and .jpg versions of the image. Tried switching to a directory name without spaces. I also tried adding the line

`image "Actions/NGS/remove_from_foil_package.jpg"`

to Klavins/Image Test. Still doesn't show up. Wonder if it is a permissions issue?

dvnstrcklnd, 2017-07-28:
"Failed to load resource: the server responded with a status of 403 (Forbidden)"

klavins, 2017-08-01:
Go to the S3 page for this image and click on "Make Public"

dvnstrcklnd, 2017-08-01:
You mean this?

[image]

Still doesn't work for me.

dvnstrcklnd, 2017-08-01:
I get: 
"Error
An unexpected error occurred."

When I try to make it public.

dvnstrcklnd, 2017-08-06:
Has this been resolved? I'm still not getting images to display.

klavins, 2017-08-06:
Ok, I think there was a problem with AWS policies. I added a new policy that should make new images be public by default. I logged in as soslab and added an image, System/Test/kubb.jpg, and it seems to have made it public. Can you try adding a new image and seeing if it is public after you add it (no need to click "Make Public")?

dvnstrcklnd, 2017-08-06:
That works, but it looks like the images is just totally public. I can open
it in an incognito using the URL. Is that what you want?

klavins, 2017-08-06:
Yeah, I was thinking these would be public. Don't really see why not.

closed 2017-08-07 by klavins