klavinslab/aquarium-bugs #68: "Cost Undetermined" error for nonexistent operations (unable to launch plan)

gasnew, 2017-07-28:
This error appears to occur in the following case:
- User builds plan with at least one unsatisfied operation input
- User removes that unsatisfied operation and **at least** the operation before it
- User satisfies the now dangling operation input (the one that used to be the output of the last removed operation) with at least one new operation.

For example:

[image]

Plate -> Transform -> Assemble. The important part of this step is that no fragments are specified (this will later result in the error).

[image]

Plate -> Transform from Stripwell. Here we satisfy the "dangling operation input" (that from Plate) with Transform from Stripwell. After clicking "Launch," we are given the Cost Undetermined error.

I've experienced this a few times before, and I'm aware of two cases this week where users have struggled with it. The best workaround I've found is to make the plan again from scratch.

gasnew, 2017-08-02:
We've encountered another bug that I think is related to this one. Essentially, in [this plan](http://52.27.43.242/launcher?plan_id=1326#), even if you remove Make Miniprep while replanning it, Make Miniprep and its predecessors will be submitted with the plan.

I think the 'x' button for removing operations while planning is somehow broken.

klavins, 2017-08-04:
This should also be fixed by the new planner GUI. Hold tight.

closed 2017-08-08 by klavins