klavinslab/aquarium-bugs #71: IWBNI the developer protocol tab autosaved

dvnstrcklnd, 2017-07-29:
Unfortunately this has become an ingrained expectation for me, especially using Mac and Google docs more.

gasnew, 2017-07-29:
While I agree this would be nice, I think it's fairly complicated to do right in practice. For example, especially when I'm teaching someone or helping them debug, we'll both have the protocol code open in our browsers. Problems could arise if my version of the protocol were to keep saving over theirs as we're working together. I also semi-frequently have multiple versions of a single protocol open in different tabs if I'm doing a major refactor. I think protocols saving my progress out of my control without other measures to ensure synchronized protocol versions can be dangerous.

What would you think about a "Do you want to save recent changes?" dialog box if you try to navigate away from a protocol you're working on? That should prevent loss of progress, and it would be more transparent to the user. klavinslab/aquarium#183

dvnstrcklnd, 2017-07-29:
I agree with everything you say here, Garrett.

dvnstrcklnd, 2017-09-11:
I think maybe this could be done in a fairly risk-free way it it only saved recovery files to the filesystem instead of trying to save to the SQL database.

bjkeller, 2017-09-11:
Also ref #192

bjkeller, 2017-09-11:
I made an umbrella issue for editor issues linked to this issue: #193, and am marking this as closed.

closed 2017-09-11 by bjkeller