klavinslab/aquarium-bugs #76: Same item number submitted for two separate tasks.

cpc3, 2017-07-31:
A user chose 1 comp cell aliquot for 2 digests so it told us to put both digests in one aliquot.  This shouldn't be able to happen.  There should be something that blocks this from being an option.

cpc3, 2017-08-10:
This just happened again and now we are having to abort the whole workflow and have the user resubmit because the technician put two digests into one comp cell aliquot so I'm making this critical.

gasnew, 2017-08-31:
I'm making a temporary solution to this on the protocol level now, but this needs to be solved on the planner/Operation Type level. One solution we've discussed is to include a flag (check box) for Operation Type inputs--something like "Single-use item" or "Item consumed." If checked, the planner would reserve the item planned with that input for use exclusively with that operation.

In the case of comp cell aliquots, we would check this box, but for many other items, such as E. coli and yeast plates, we'd leave it unchecked in most cases.