klavinslab/aquarium-bugs #82: Upload spreadsheet doesn't ignore Byte Order Mark

jvrana, 2017-08-02:
I frequently get this error:
` "Error processing spreadsheet: Could not find sample type ï»¿SampleTypeName"`

 This mark **ï»¿** is a byte order mark (I guess) that I think the upload spreadsheets should ignore. Its present when I save a csv through excel or through any of my text editors. Its pretty difficult on my end to remove it.

For ruby, I believe you add the "r:bom|utf-8" tag to fix this?

```
File.open('file.txt', "r:bom|utf-8"){|file|
  ...
}
```

klavins, 2017-08-03:
Can you send me a csv file that causes this problem so I can test it?

jvrana, 2017-08-03:
Hmmmm, ok I sent it via slack, however in the process of sending it, the error disappeared after I redownloaded it from slack and tried to upload it.

klavins, 2017-08-04:
Gotta love those one line fixes. In this case, it was reading the file as text in javascript instead of as a binary string. Seems to work on your test files and other test files that I've bene using.

closed 2017-08-04 by klavins