klavinslab/aquarium-bugs #96: No follow-up for making plasmid glycerol stocks

dyounger, 2017-08-04:
I am still really confused by the way glycerol stocks are made. After an upload sequencing result is done, there is a data field where I can type in "yes" or "no." Do these actually do anything? From what I can tell, they don't. That is really misleading and it is VERY important that glycerol stocks are made. If they only do anything when a glycerol stock protocol is manually planned, that has to be very clear.

klavins, 2017-08-07:
The message in the user response field is context dependent, but maybe should not be.

ayeshasal, 2017-08-11:
what the "yes" "no" do is, IF you've planned a glycerol stock, the protocol will instruct the tech to make a glycerol stock, throw out the plate, and throw out the overnight depending on the user's answer. you're right, though, i can see how that's super confusing.

closed 2017-09-21 by gasnew