klavinslab/aquarium-bugs #98: Saving after mark_as_deleted

dvnstrcklnd, 2017-08-05:
Is there ever a case where you would *not* want to save after calling mark_as_deleted on an item? It would be nice if the method saved automatically. As it stands you need:
```
operations.each do |op|
  lib = op.input("Yeast Display Library").item
  lib.mark_as_deleted
  lib.save
end
```
when you could just do: 
`operations.each { |op| op.input("Yeast Display Library").item.mark_as_deleted }`

This may also apply to other item methods.

gasnew, 2017-08-07:
I made a minrep of this in "Minrep->mark_as_deleted test" on the nursery. It seems to work fine without .save now. Is this what you're talking about?

dvnstrcklnd, 2017-08-07:
I don't know, I was responding to the documentation, which says:

"i.save - if you make changes to an item, you have to call i.save to make sure the changes are saved to the database."

I was also looking at someone else's code which used the three line structure with the temp variable. 

I'm not sure where the deleted flag shows up in the DB, so I can't tell whether your minrep works.

It isn't really a bug. So I'm removing that label.

dvnstrcklnd, 2017-08-07:
In item.rb there is:

```
def mark_as_deleted 

    write_attribute(:location,'deleted')
    self.quantity = -1
    self.inuse = -1
    self.locator_id = nil
    locator.item_id = nil if locator

    r1,r2 = [false,false]

    transaction do
      r1 = self.save
      r2 = locator.save if locator
    end

    r1 && r2

  end
```

Which makes me think the documentation is wrong. Let's discuss it at the meeting tomorrow.

closed 2017-08-11 by ayeshasal