klavinslab/aquarium-bugs #104: Discard items doesn't work

dyounger, 2017-08-07:
The discard items plan doesn't work. It is also very clunky as it is now. I would like to be able to have a choice of either selecting specific item numbers OR giving an object ID and then selecting whatever items I want to get rid of for that object. It should also be possible to easily delete multiple items in a single plan.

klavins, 2017-08-07:
I wonder if the sample should be the input to this protocol so then the user would choose which item when they fill in the sample name?

klavins, 2017-08-07:
Might also be an array input top the user only has to submit one plan.

ayeshasal, 2017-08-11:
i think we can do array inputs for this -- that way, you can delete specific items for a single object type by inputting all the sample ids of that object type you wanna get rid of.

gasnew, 2017-08-29:
Justin's Discard List of Items should address all this! I've updated it to include an optional array input as well.

closed 2017-08-29 by gasnew