klavinslab/aquarium-bugs #105: Some temporary items aren't being deleted

dyounger, 2017-08-07:
Ex. for gel purify plan, sometimes the 50mL .8 percent gel in gel box is left over, even though it is clearly not still around because it was cut and then the gel slice was purified. This is particularly annoying because I can't delete a collection, so there is no way for me to get rid of it.

gasnew, 2017-08-09:
Could you please show me an example? The Extract Gel Slice protocol is written to delete the gel boxes at the end.

dyounger, 2017-08-11:
Sorry about the slow reply...

Yes, en example is for fragment 20482 (but there are many more). It isn't happening with all of them, though, so it is possible that this is just because a protocol crashed or something before they were deleted.

gasnew, 2017-08-23:
That's a good point. I'm working with Cami and Sam on this on the manager side. And I think the Discard Items operation (the one Justin made) will address this further once we close #104.

closed 2017-08-23 by gasnew