klavinslab/aquarium-bugs #109: Plan Templates for GUI Planner

jvrana, 2017-08-08:
I'm assuming this is a coming feature since it seems obvious and we've been talking about it for awhile.

I was a little disappointed when I launched a plan in the new graphical planner and it disappeared from the Planner. For some reason I thought it'd stick around in a "Saved" plan tab on the left and I could just re-submit it repeatedly with different samples as outputs. I guess these would be "plan templates" rather than plans.

jvrana, 2017-08-08:
Anyways, its pretty easy to replan and relaunch the same plan. Making this a convention in the new gui planner would be pretty rad. Especially with the entire cloning workflows saved.

klavins, 2017-08-08:
Yes, templates are coming. They will be a little different, in that you will be able to either "make new from template" or paste the entire contents of a template into a plan you are editing. We'll also make it so that you can share templates with other users (if you are an admin).

But tight now in the planner you can copy a plan and rename it something like "My Template". Then you can copy it whenever you want to use it and then launch it. Also, as you say, you can send a plan from your plan list back to the planner to re-edit.

closed 2017-09-12 by ayeshasal