klavinslab/aquarium-bugs #110: Module Blocks in GUI Planner

jvrana, 2017-08-08:
Probably another obvious feature, but it'd be cool to be able to import save plans (or plan templates?) into your existing plan. This way saved plans could function not only as stand-alone plans that could be resubmitted with different samples, but also as functional blocks that you import and build upon in your plan.

E.g.
  Plan 1: Miniprep Block
  Plan 2: Sequencing Block
  Plan 3: I create a new plan an import Plan 1 and Plan 2, wire them together and save for make a new workflow.

jvrana, 2017-08-08:
I guess something that could be different is if a plan is a new "block", a box goes around it and the inputs and outputs and 'pushed' to the surrounding block so you can treat many operations as a single functional block

klavins, 2017-08-08:
Yes that's coming soon.

closed 2017-09-13 by Gamemackerel