klavinslab/aquarium-bugs #114: Graphical Planner: Building operations 'upwards' doesn't faithfully keep object types

jvrana, 2017-08-08:
If you build an operation upwards, the object type of the input of the new operation just defaults to the first allowable field type instead of the output object type of the operation you were building upon.
Below, is an image which I put a "Trypsinize Plate" from a "Plate Cells."
[image]

The orange wires are very nifty though for indicating mismatches!

klavins, 2017-08-08:
Well, at least it highlighted the wire connecting the two inconsistent object types :-)

bjkeller, 2017-09-20:
@jvrana Can you confirm whether this is still an issue with new planner?

emernic, 2018-02-28:
Not super urgent since the arrow does show up orange, but this seems to still be an issue as of commit 0cf3253a5f6faad7859274278e4ead2b2516e994 @bjkeller 
Noticed this bug with a user yesterday.