klavinslab/aquarium-bugs #116: Graphical Planner: Long operation name displays incorrectly

Gamemackerel, 2017-08-08:
It looks like the expected behavior In order to display operations names on one line, any name that is too long to display on one line is cut short and appended with ellipsis.
It's a little hard to see what's happening in the image but for my browser the "make Electrocompetent Cells (JV Optimized)" operation name is handled wrong - for some reason the ellipsis wants its own line.

[image]

Chrome Version 60.0.3112.90 Windows 10

Gamemackerel, 2017-08-08:
Trimming 3 more chars should do the trick
[image]

klavins, 2017-08-08:
Its because your browser is adding that scrollbar. I've had these scrollbar issues before and I think I know how to fix them.

closed 2017-09-11 by Gamemackerel