klavinslab/aquarium-bugs #122: Users are being incorrectly charged for glycerol stocks.

cpc3, 2017-08-09:
If a user says their sequencing is incorrect, the plate and stock are thrown out, but they are still being charged for a glycerol stock even though one isn't made.

Gamemackerel, 2017-08-09:
Not sure how the cost service works, so I hesitate to make any changes. 
Can this be fixed simply by making this addition to the each loop in "Discard plasmid stocks for no" block?

<pre>  
    show do
        title "Discard Plasmid Stocks from bad sequencing results"
        
        note "Please discard the following Plasmid Stocks:"
        operations.select { |op| op.temporary[:no] }.each do |op|
            stock = op.input("Plasmid").item
            note "Plasmid Stock #{stock.id} at #{stock.location}"
            stock.mark_as_deleted
            stock.save
            <b>op.error :bad_seq, "Bad sequencing results."</b>
        end
    end if operations.any? { |op| op.temporary[:no] }
</pre>

ayeshasal, 2017-08-10:
yeah, i think that should fix it! nice. i'll check to make sure but i'm closing this issue for now

closed 2017-08-10 by ayeshasal