klavinslab/aquarium-bugs #139: You can add two overlapping wires in graphical planner

jvrana, 2017-08-15:
You can add two overlapping wires in the graphical planner. In the example below, a use should not be allowed to create another wire from Packaging Plate output to the Harvest Lentivirus input since this wire already exists. Currently you can create another overlapping wire by shift-click; you can tell there is more than one wires by selecting and deleting the 'top' wire, while the wires below remain.

[image]

bjkeller, 2017-09-11:
Rephrasing: if a wire exists from an output to an input, then a new one should not be added.

ayeshasal, 2017-09-12:
#127 was about a similar issue, except in that case the issue was that there were multiple outputs being wired into one input, which should not have been allowed.

closed 2017-09-12 by ayeshasal