klavinslab/aquarium-bugs #142: Can't make glycerol stock of unverified plasmid

dyounger, 2017-08-15:
It seems like the glycerol stock task still requires a verified plasmid (or a verified sequencing result) in order to make a glycerol stock. It is starting to become really difficult to tell which of my stocks are good and which aren't. This is even more difficult because there is no discard item, so there are a lot of stocks that I have to go through and manually delete to make sure that only the right one is left.

jvrana, 2017-08-16:
This doesn't solve this glycerol stock problem, but there is a Discard Item in the Misc category. However it seems lame since you have to submit a single item for each discarded item.

There's a new **Discard List of Items** operation in the Cloning that should make it easier for you to discard items. It takes a comma or space separated list of item ids and discards them. You'll get a warning if (1) you don't own the sample the item is associated with, (2) item is already discarded, (3) item doesn't exist. 

Items that do exist will appear as inputs in the operation, which will be viewable in the planner or manager view.

Examples of valid parameter inputs are:
`165332 154223 64332 1343  434`
`165332, 154223, 64332, 1343,434`
`165332, 154223, 64332,    1343   434`
Example of invalid inputs are:
`165332, DAVID, `
`   `
`15Agca`

dyounger, 2017-08-16:
Ok, cool! I will try this out.

closed 2017-09-21 by gasnew