klavinslab/aquarium-bugs #153: Check Divided Yeast Plate Minor bug

wmvoje, 2017-08-22:
When attempting to make a divided yeast plate from a glycerol stock, the "Streak Plate" box input and outputs Nodes fail to both turn green. 

How to replicate:
1) Input "Check Divided Yeast Plate"
2) ADD "streak plate"
3) On "Streak Plate" input node select "Suspension Yeast Strain / Yeast Glycerol Stock"
4) Enter yeast strain number in the IN or OUT text field entry. 
5) Whichever in/out were selected for the entry is now green with the other being orange - However both of the text fields are green

The remaining orange node can be made green by manually entering the strain number again in the respective text box. Or by saving the Plan.

Gamemackerel, 2017-09-11:
I think this is the same issue as #117

wmvoje, 2017-09-11:
It's very similar. Not sure if it's identical.