klavinslab/aquarium-bugs #157: Operations can appear out of view in the planner

jvrana, 2017-08-22:
You can add operations to be out of view of the planner, which the user is unable to control. At least one user has complained about this.

MinRep: You can recreate this by adding a successor near the top of the planner as in the picture below.

[image]

klavins, 2017-08-24:
Note that for now you can press A (or ctrl-A in the next version) to select all operations and move them into view.

bjkeller, 2017-09-05:
@jvrana What is the desired behavior?  That a new operation is only created within the current view, or that the view changes?

jvrana, 2017-09-06:
I think if the view got taller as you add operations out of view. But either of those solutions would be fine.

bjkeller, 2017-09-11:
I have a commercial tool that allows you to create a graph that is described using an outline. It creates new nodes in the graph as you type into the outline. It also will create nodes outside of the working view, but the view can be scrolled and resized. Automatic layouts can also be applied.