klavinslab/aquarium-bugs #169: If a primer is used twice it's checked for contamination twice and remade twice if contaminated

cpc3, 2017-08-30:
In the PCR step 17 it asks the tech to check primer "76769" twice.  And when they said it was contaminated they are asked to make two new primer aliquots in the next step, not just one. 

http://52.27.43.242/krill/ui?job=44437

Gamemackerel, 2017-08-30:
This is probably a result of my updates to the volume checking part of the
protocol last week, I'm on it

Gamemackerel, 2017-08-30:
It seems for this PCR batch, some aliquots have been selected to be used simultaneously for multiple operations, but they are single use items as far as I am aware so the volume checker method does not account for this.

Operation 11737 from plan 2259 and operation 12027 from plan 2282 share the same reverse primer (96224)

Operation 11744 from plan 2260 and operation 12037 from plan 2282 shared the same forward primer (76769)

When 76769 was marked as contaminated for each operation it was in, it was deleted and the protocol conveniently had the tech make two new distinct aliquots, so the operations were probably completed successfully each with their own unique aliquots.

Unfortunately there is currently no way to prevent a user from planning one single-use item to be used in multiple operations, this might be a nice feature. 
Fortunately, make PCR has the tech make new aliquots on the fly for any that are marked as low volume! If a tech comes across an operation that wants to use one aliquot for two or more operations, they can simply mark all primer aliquots with duplicates as low volume or contaminated and the protocol will task them to make new unique aliquots of the same sample type for those operations.

Operation 12022 from plan 2282 which hasn't been run yet is also is set to use 76769 as the fwd primer, even though it is marked as deleted. 
When this operation is run: as long as the tech marks aliquot 76769 as low volume, they will be told to make a new aliquot of the same sample type from stock and be able to proceed normally.