klavinslab/aquarium-bugs #193: As a Workflow Developer, I want to be able to use the Aquarium protocol editor without losing my work

bjkeller, 2017-09-11:
Developers report losing work when editing in the protocol editor due to:
1. leaving the page without saving (see #192 )
2. having the session closed by visiting another Aquarium server

Issue #71 (autosaving) is related to both.

bjkeller, 2017-09-16:
@Gamemackerel @gasnew or @ayeshasal You should be able to check out some of the behavior on your local machine by pulling the branch `developer-refactor` and then running your local server.  However, the problem from switching between nursery and production can't be tested this way.

Gamemackerel, 2017-09-18:
Hey Ben,
I know the session logout fix is pushed to nursery. Is it not supposed to work until the code is pushed to both servers?
It is still not possible to have two servers of aquarium open at once, and opening production still logs you out of nursery and vice versa.

bjkeller, 2017-09-18:
I see there is one more cookie. I'll look into it.

bjkeller, 2017-09-18:
I don't think we'll be able to fix this immediately. The stray cookie is hard-coded into  the [angular_rails_csrf](https://github.com/jsanders/angular_rails_csrf) gem even though the Angular API allows the cookie name to be configured.  We either need to replace the use of the gem or make a PR and see if they'll accept it.

bjkeller, 2017-09-19:
Closing this issue and pushing unfinished issue to #230.

closed 2017-09-19 by bjkeller