klavinslab/aquarium-bugs #206: minor Planner UI Glitch: wires turning orange when they shoudn't for array inputs

Gamemackerel, 2017-09-13:
For wires to array inputs that have correct object/sample type match, they turn orange until filled in with a sample.
[image]

Gamemackerel, 2017-09-13:
Sam was experiencing other glitches with orange wires for array inputs. I believe for him sometimes wires would be orange even when the output of Purify Gel Slice was connected to an input for Assemble Plasmid and both were green and filled in.  I could not reproduce his issue.

klavins, 2017-09-13:
Ok, I've updated how this is computed and hopefully it will make sense. Basically, now the object/sample have to match and the samples chosen (if present) has to match. If one or both i/o pins don't have samples, then the wire is not turned orange.

closed 2017-09-13 by klavins