klavinslab/aquarium-bugs #219: As a lab manager, it would be nice to see item numbers for all inputs for all pending operations

gasnew, 2017-09-15:
## Bug report

- **Reproduce by:** *Submit a plan with two operations (e.g., Check Plate -> Make Overnight Suspension. [See this plan](http://52.27.43.242:81/launcher?plan_id=47384).)*
- **Expected:** *Plate item id should show up in the input row on both Planner and Manager view.*
- **Actual:** *No item id appears.*
- **Acceptance criteria:** *The item id from the previous operation's output should be associated with the succeeding operation's input upon completion of the previous operation.*

gasnew, 2017-09-15:
It's worth noting that this already works correctly for operations that are leaves in their plans and for operations that have been scheduled, errored, and set back to pending.

klavins, 2017-09-15:
I have added this functionality. Previously operations.retrieve did this work. Now, when an operation is moved to pending, it looks for predecessor outputs and copies the item and row and column (if any) to the input. 

operations.retrieve still does this too (no harm as far as I can tell).

gasnew, 2017-09-15:
This is great; thanks!

closed 2017-09-19 by klavins