klavinslab/aquarium-bugs #221: Planner GUI glitch: Module outputs does not hook up to array inputs

Gamemackerel, 2017-09-15:
When the output of a module that produces a fragment stock is wired in as an input for assemble plasmid, propagation fails in either direction, and assemble plasmid refuses to use the output of the module as it's input, either marking itself orange or using already made inventory of that type.

klavins, 2017-09-15:
I can't reproduce this problem. Here's what I tried: A made a module with "Purify Gel" in it. I hooked its output up to the input of "Assemble Plasmid". I tried changing the sample for each end of the wire and it propagated and changed the colors correctly.

klavins, 2017-09-15:
[image]

Gamemackerel, 2017-09-15:
Yup I can't reproduce it now either. I think the issue was caused from using modules that were saved from a previous version?

klavins, 2017-09-15:
Sometimes it might have to do with whether the plan has been saved or not, or whether it is the first plan you've made since reloading the page, or not, etc.

Gamemackerel, 2017-09-15:
I found the issue behind this, it is pretty significant. 
It looks like the issue occurs for operations that are included in a template without pre-specified input or wiring. When a template with an operation like this is loaded, the inputs for the operation behave oddly. the inputs will only accepts existing inventory items, or wiring from adding suggested predecessors. Shift clicking from a prior predecessor to the input fails to create a wire, and shift clicking from a compatible module output creates a 'false' wire

Below on the left, the check plate operation was grabbed from the operation list, the propagation and wiring works just fine. On the right, the check plate operation was pulled from a singleton template, and its input will not allow wiring to the predecessor, with shift+click. 
[image]

Gamemackerel, 2017-09-15:
The fact that I thought this was specific to array inputs / module outputs arose from a coincidence in how I was testing. I was testing with an assemble plasmid operations that I got from a template, but testing non-array input operations from the operations list. Then I hooked up the module to the Assemble Plasmid with the faulty right click, but I hooked up an operation to the assemble plasmid using the suggested predecessor button, which does work in this case.

klavins, 2017-09-15:
I think I have reproduced the issue. One thing I noticed is that if you paste the template, save the plan, and then wire things up, it works. I think. Anyway, working on it ...

Gamemackerel, 2017-09-15:
Yea saving it makes it work fine, so this probably isn't quite as dramatic of a glitch as I made it sound in my email

klavins, 2017-09-15:
Ok, found it. It was a one line fix (removed the computation of a nonexistent getter). Let me know if you see similar issues with pasting templates.