klavinslab/aquarium-bugs #222: Cannot link the input to an output within the same module

ayeshasal, 2017-09-15:
this is relevant to yeast qc, where you want to link the input plate to the output (so, overall, the plan looks like this: qc --> overnight --> glycerol stock; qc takes in a plate, and we want the same plate to be an output, but none of the operations within qc (yeast lysate, PCR, fragment analyzing) output a plate, so i want to hook the input of the module to the output).

klavins, 2017-09-15:
A workaround for now would be to give qc another output, the plate. Then take that output and route it to the output of the module.

ayeshasal, 2017-09-15:
yup that's what i'm doing right now, but i wanted to bring this up just in case there were cases in other workflows where this might be useful. if the yeast is the only place where this happens, though, it might not be worth fixing.