klavinslab/aquarium-bugs #224: Propagation of input into multiple operations doesn't work in modules

ayeshasal, 2017-09-15:
- **Reproduce by:** creating a module with one input linked to two different operations, e.g., link the input of module "Yeast QC" (which i saved as a system-wide template) to two "Yeast Lysate" operations and then input a sample id
- **Expected:** the sample should propagate into both yeast lysate operations
- **Actual:** it doesn't -- everything's blank
- **Acceptance criteria:** when everything is filled out correctly, i.e., the inputs/outputs of all the operations within the module have been filled out with the correct sample ids

klavins, 2017-09-18:
Ok, I have added this as a special case of propagation. Hopefully I didn't break anything in the process. The fix gets right into the heart of the propagation code.

ayeshasal, 2017-09-18:
awesome! i'll test it out and let you know if it works.

ayeshasal, 2017-09-18:
i'm still having this issue. inside the module, this is in the left set of protocols: 
[image]
and this is in the right set of protocols: 
[image]

the input for the entire module is sample id 16666

klavins, 2017-09-19:
I am not sure I know what you mean by left and right set of protocols. Can you just save a plan your "Your Plans" and tell me which one it is and how to break it?

klavins, 2017-09-19:
Is this still a problem?