klavinslab/aquarium-bugs #226: Nursery sign-in page doesn't work

Gamemackerel, 2017-09-18:
I am unable to access the sign in page for nursery. Is this just me? Possibly caused by the fix for the cookie issue #193
[image]

Gamemackerel, 2017-09-18:
Here's that error message:
>angular.js:4987 Uncaught Error: [$injector:unpr] http://errors.angularjs.org/1.6.6/$injector/unpr?p0=aProvider%20%3C-%20a
    at angular.js:88
    at angular.js:4826
    at Object.d [as get] (angular.js:4981)
    at angular.js:4831
    at d (angular.js:4981)
    at e (angular.js:5006)
    at Object.invoke (angular.js:5032)
    at angular.js:4842
    at p (angular.js:410)
    at hb (angular.js:4842)

bjkeller, 2017-09-18:
It could be either one.

Looks like I messed up an injector. 

Changing the html tag of `aq2.html.erb` to 
```
<html ng-app='aquarium' ng-controller='layoutCtrl' ng-cloak style="overflow: hidden" ng-strict-di>
```
causes this to happen on local.

bjkeller, 2017-09-18:
The problem was in the injector for the onbeforeunload handler.  The current version of master resolves this, but I'm having an issue with already having a new library in the db.  I'm not sure if this Is a real problem or not.

klavins, 2017-09-18:
Ben, The reason this doesn't work on nursery is because nursery uglifies the js. So you can't do

  w.controller('codeCtrl', function (  $scope,   $http,   $attrs,   $cookies ) {
     ...
  });

but have to do

  w.controller('codeCtrl', [ '$scope', '$http', '$attrs', '$cookies', 
                  function (  $scope,   $http,   $attrs,   $cookies ) {
    ...
  }]);

klavins, 2017-09-18:
Ah. I see you fixed it on master. I have pushed this code to nursery and am able to log in.

klavins, 2017-09-18:
What is this issue "having a new library in the db"?

bjkeller, 2017-09-18:
A library with the default name is already in the database, and so creating
a new library fails.

bjkeller, 2017-09-18:
@Gamemackerel if this is working, please move it to done.