klavinslab/aquarium-bugs #229: Propagation doesn't work properly with modules when linking 3 modules to one operation

ayeshasal, 2017-09-18:
### **Reproduce by:** 

if you go into nursery and link up 3 of the "Sequencing | 1 Primer" modules to make miniprep and then edit the primers/plasmids, you get some really weird behavior. 

### **Actual behavior**

assume you've set up your plan like this: 
[image]

**A** refers to the leftmost sequencing, **B** refers to the middle sequencing, and **C** refers to the rightmost sequencing. 

the plasmid forward propagates properly (like, if i change the output plasmid for make miniprep, all 3 sequencings have the same plasmid) but if i change one of the input sequencing plasmids (i.e., i edit **C**'s input plasmid), only that one, **B**'s input, and the miniprep's output are changed -- **A**'s input plasmid field becomes blank.

this issue also happens with primers, which is a little more alarming because users are expected to have different primers for all 3 sequencings. we see similar behavior: if i change **A**'s primer, **B**'s primer is changed to be the same and **C**'s primer is blank. a similar thing happens when i change **C**'s primer. if i just change **B**'s primer, though, **A** and **C** aren't affected. 

### **Expected:** 
changing the primer on one sequencing shouldn't affect any of the other primers and changing the plasmid (either as an output for miniprep or an input for the sequencings) should propagate properly.

klavins, 2017-09-18:
I do not get this behavior on nursery when I do this. 

I am getting some strange behavior when you choose a primer, navigate away and then back, the primer name is gone and replaced by just what you typed for autocomplete.

ayeshasal, 2017-09-18:
that's so weird...i'm still getting this behavior (though when you navigate away and then back, the primer name has been replaced with what you typed for autocomplete for me, too). 

also, i've gotten some more bugs with propagation inside/outside modules -- i think i'm gonna make an "epic" for that.

klavins, 2017-09-18:
Ok, Now I am figuring it out. It changes the sample of the first sample you change, no matter what sample you change. I think. I'm getting close to figuring this one out.

klavins, 2017-09-18:
I think I got it. I had a closure that i didn't think was a closure. Fixing a variable scope solved the issue, hopefully. Please check.

ayeshasal, 2017-09-19:
i'll check it as soon as i get home!

ayeshasal, 2017-09-19:
it works yay!!

closed 2017-09-19 by ayeshasal