klavinslab/aquarium-bugs #234: sequencing batch download

dvnstrcklnd, 2017-09-19:
As a user, it isn't really useful to me to download a batch of sequencing files if they contain a bunch of sequences that don't belong to me. And even though none of the data is really secure yet, I don't like that the software downloads other users' data to my computer when I am trying to retrieve my own data.

dvnstrcklnd, 2017-09-19:
To clarify re the title, this capability already exists as long as you download the files one by one. However, it also gives an option to batch download, but this appears to download the zip file provided by GeneWiz, which contains many users sequences batched together. Thinking about it some more, the "security" aspect is probably more important than the convenience aspect: To the uninitiated user, the behavior seems buggy and insecure, and only achieves limited utility.

However, there are a number of issues with this interface that are probably best taken together.

bjkeller, 2017-09-20:
The issue is that the zip file is the one provided by GeneWiz and not constructed for the user's samples.