klavinslab/aquarium-bugs #253: As a Trident API User, the api is not returning protocol code correctly

jvrana, 2017-09-28:
## Bug report

- **Reproduce by:** requesting any code through the json api. For example `print(Code.find(965).content)` in Trident from the nursery sever returns the default protocol code instead of the code defined in the developer tab
- **Expected:** should return the code located in the OperationType
- **Actual:** returns the default protocol code instead of the acutal code.
- **Acceptance criteria:** For example, `print(Code.find(965).content)` should return `#IT WAS FIXED!` in the nursery server

bjkeller, 2017-09-28:
You can get to the code for an operation type.

I don't think we should allow arbitrary queries like this from outside of
aquarium. It will just make it harder to improve internal organization.

jvrana, 2017-09-28:
Right, I see. Thanks.

closed 2017-09-28 by jvrana