klavinslab/aquarium-bugs #259: As a pydent user, it would be helpful if there was an efficient back-end method for finding primers by full sequence

dvnstrcklnd, 2017-10-10:
As a pydent user planning cloning workflows, I want to check if a primer exists in the inventory before I order a new one. This is challenging, because the sequence is split into two field values.

I wrote this python code to do the search, but it takes about 10 seconds to execute for a single primer:

```
def find_by_sequence(anneal_seq, overhang_seq=""):
     full_seq = overhang_seq + anneal_seq
    
     overhang_seqs = []
     anneal_seqs = []
    
     for i in range(len(full_seq) + 1):
         overhang_seqs.append(full_seq[:i])
         anneal_seqs.append(full_seq[i:])
    
    overhang_fvs = aq.FieldValue.where({"name": "Overhang Sequence", "value": overhang_seqs})
    anneal_fvs = aq.FieldValue.where({"name": "Anneal Sequence", "value": anneal_seqs})
    
    if overhang_fvs and anneal_fvs:
        overhang_parents = {fv.parent_sample.id for fv in overhang_fvs}
        anneal_parents = {fv.parent_sample.id for fv in anneal_fvs}
        matching_sample_ids = list(overhang_parents.intersection(anneal_parents))
        if matching_sample_ids: return aq.Sample.where({ "id": matching_sample_ids })
  ```

I think that if this code were executed in Ruby at the back end it would go much more quickly because the limiting step seems to be data transfer in the FieldValue.where queries.

jvrana, 2017-10-11:
This part of your code seems like it would take a very long time:

```python
    for i in range(len(full_seq) + 1):
         overhang_seqs.append(full_seq[:i])
         anneal_seqs.append(full_seq[i:])
    
    overhang_fvs = aq.FieldValue.where({"name": "Overhang Sequence", "value": overhang_seqs})
    anneal_fvs = aq.FieldValue.where({"name": "Anneal Sequence", "value": anneal_seqs})
```

Its doing 2*(1+2+3+...+n) queries in this step. I think it would be better to grab all the field values that are named "Overhang Sequence" or "Anneal Sequence." Create a table (maybe internally in python) that match parent_sample_id and concatenate the sequences, and then search for your sequence using that table (either regex or other).

Even better would be to pull FieldValues with a JOIN on parent_sample_id, but I'm not quite sure how to do that at the moment.

jvrana, 2017-10-11:
Additionally, I believe the `fv.parent_sample.id` is doing an additional query that is unnecessary. I think you can just use `fv.parent_sample_id` instead to avoid doing additional queries.

dvnstrcklnd, 2017-10-11:
Interesting. I actually did it that way because `fv.parent_sample_id`
doesn't work. I didn't understand why, but it is because it has to be
`fv.parent_id`. It still takes way too long to query.

bjkeller, 2017-10-11:
The problem with providing the solution is that you can jump past the
actual problem.

You actually want to match a single primer sequence, right?

Please give an example of the call you would like to make.

dvnstrcklnd, 2017-10-11:
If data transfer is the limiting step, then doing an SQL query that grabs
as little data as possible is probably the right way to go. The list of all
the field values that are named "Overhang Sequence" or "Anneal Sequence" is
huge. Even a pretty hairy SQL query would be much faster than that data
transfer.

In this case, I think the JOIN would result in transferring a lot of data
that isn't necessary. I just need to be able to get an intersection of
parent_ids.

But if this was a back-end query, you may be right.

dvnstrcklnd, 2017-10-11:
> issuecomment-335938958
> z526LldPx6AAqbks5srSXagaJpZM4P0gll
> 335949226>,

I don't think this jumps over the problem in the context of the existing
data structure. You have to do something like the query I am proposing to
exhaustively search the sequences. If you are going to make the change we
discussed this morning, then I agree it jumps over a better solution.

dvnstrcklnd, 2017-10-11:
I'm closing this anyway because we've more or less decided to change how the primer sequences are stored.

closed 2017-10-11 by dvnstrcklnd