klavinslab/aquarium-bugs #264: Urgent: protocol deployment delay

Gamemackerel, 2017-10-24:
The most current version of a protocol as visible in the "protocol" tab of an operation, is not availalble to immediately be run on either tester or launched operations. 
After saving a modification to the protocol code, for at least 5-10min the operation still uses the old version on both production and nursery.

Gamemackerel, 2017-10-24:
Sarah is able to replicate on several of her protocols, including '24 well growth', but this does not happen on every protocol.
Every time she does an modification, the current version in the _versions_ page displays the most recent update, but the code that is deployed is the n - 1 version.

This (I think) is the same issue as #262

bjkeller, 2017-10-31:
@klavins 
A protocol that we are having trouble with is "Induction - High Throughput": "24 well inducer plates".
I believe if you look at it at the moment, there is a dummy show block at the top of main that says "hi3".  If you remove the show block and run test, you will still get the message.

I'm leaning toward this being an operation gone bad, because when I attempted to make a copy, I got a console error related to the current_operation_type.model being undefined.
However, on reload the copy seems to have been made correctly, and the test generates the correct output. See "ZZBen": "24 well inducer plates (copy)"

bjkeller, 2017-10-31:
those are on nursery

closed 2017-11-13 by Gamemackerel