klavinslab/aquarium-bugs #270: Export of protocol followed by Import adds weird "A-hat" char before "mu"

segoldberg, 2017-11-14:
See Parameter Name below:
[image]

## Bug report

Export of protocol (from nursery) followed by Import of .json (into production) adds weird "A-hat" char before "mu"

Reproduce by: 
Export the following protocol from nursery:
Induction - High Throughput/Flow Cytometry 96 well (copy)
Import into production.

bjkeller, 2017-11-14:
As a workaround, when you want a symbol to be displayed in a protocol use the [html character references](https://dev.w3.org/html5/html-author/charref). In this case, I would use `&micro;` for &micro;L. Since a show is being rendered as HTML this works fine.

bjkeller, 2019-06-03:
Fixed by https://github.com/klavinslab/aquarium/commit/8fe5aa48cc5644f981a710abd7b785b02f5e0efd

closed 2019-06-03 by bjkeller