klavinslab/aquarium-bugs #272: feature request: upload button remain "live" until used

segoldberg, 2017-12-04:
## Feature request
As a technician, I want the upload button to be "live" even if I skipped over it (by mistake, or to read the next few protocol steps). It would be better if it turned gray only after it was used for uploading.

farheensal99, 2017-12-18:
Hi! Could you provide more information on which upload button? A screenshot would be great. Thanks!

segoldberg, 2017-12-18:
THE PROBLEM: for a show block containing the "upload" command, if you press
the green ok (or right arrow) before uploading, the upload button is no
longer active.
THE REQUESTED FEATURE: keep the upload button live as long as nothing has
been uploaded, or if the technician goes back to that step.

i do not have screenshots.
any code containing the "upload" command. example:

uploads = show do
                title "Select a file"
                upload var: "uploaded_file"
end

for testing, you can use the highlighted protocol (on production, make your
own copy) to request and up load to existing items/plans.
[image: Inline image 1]