klavinslab/aquarium-bugs #291: Bug: No longer able to create new Libraries (Nursery)

jvrana, 2018-03-23:
**Reproduce by:** Create a new Library on the developer page. Then update the Library.
**Expected:** The Library is created and you can edit it.
**Actual:** Aquarium immediately deletes the new library and logs you out.
**Acceptance criteria:** You can create, edit and save Libraries in the developer page.

klavins, 2018-03-23:
I can't reproduce this. It may be that you are not logged in. Aquarium has a bug where you can be logged out but still see the page you were on. But when you try to do something that makes a background json request, it fails because you are not logged in. Could that be it?

jvrana, 2018-03-23:
I biked home and its working now. I took a look at the json requests it was making and didn't see anything weird. Meanwhile, logging into production, things were working fine. I guess I'm going to close this and the other issue.

closed 2018-03-23 by jvrana