klavinslab/aquarium-bugs #294: Bug: More strange interactions between nodes of unconnected modules in planner

emernic, 2018-04-06:
## Bug report

- **Reproduce by:** *Add in ~20 modules to a plan from templates.*
- **Expected:** *Modules function independently unless connected by wires.*
- **Actual:** *Nodes of separate modules are frequently linked (even between different modules). This occurs seemingly at random and has been observed in almost every plan where the user attempts to use many modules together.*
- **Acceptance criteria:** *Modules function independently unless connected by wires.*

[image]

bjkeller, 2018-04-06:
I think we really need to get you on trident...

bjkeller, 2018-04-06:
> @bjkeller <https://github.com/bjkeller> Trident definitely would be
> helpful for dealing with plans that are impractical to assemble w/ drag and
> drop interface. Is it ready for this? I can download and try it out.
> However, I suspect that the vast majority of our day-to-day users are not
> going to want to use anything other than the GUI.

Devin is using it to build complex plans including based on the puppeteer
output you guys have sent. Though, I wont guarantee that you wont run into
issues.

If it is possible to hide some details, maybe you could construct a plan
with trident and create a module to expose to users.

Regardless of whether the graphical planner was perfect or not, I think
these complex plans need a different way to indicate input.  We have at
least one example of that locally.

emernic, 2018-04-06:
@bjkeller 
> a different way to indicate input

Not sure that I am understanding what you mean by this.

We will need someone in our group with a working knowledge of Trident anyway, so I will try it out.

bjkeller, 2018-04-06:
It is not clear that it would be applicable in your case, but our local
example would likely involve tabular input.  With the graphical planner, if
the plan has repetitive structure it is easy to make mistakes.

emernic, 2018-04-06:
@bjkeller 
I see. Yes, some way of reducing errors for large repetitive plans would be very helpful. However, it would be nice if we could get some of the bugs with the existing planner ironed out soon. Planner issues are the main remaining hurdle in convincing users to submit jobs to us. If these were fixed we could probably push the percentage of our building's cloning that goes through Aquarium from ~20% to ~70%.

bjkeller, 2018-04-06:
Understood.